AWSW-de/WordClock-16x16-LED-matrix #1: Use Arguments for ew command Server

reptil1990, 2023-01-09:
Hey, 
Just finished the first build and it is awsome.
One thing would be very usefull.

If you set an ew command with the 8080 webserver it is a toggle. It would be nice to have an argument behind the command to set the status. So you can get a defined state.

Example: 
http://192.168.1.102:8080/ew1?on=1   -> to turn the ew1 command on 
http://192.168.1.102:8080/ew1?on=0   -> to turn it off.

Makes setting the command much more reliable in a home automation.

Thanks for your work!

AWSW-de, 2023-01-09:
Hello,
thanks for your nice words and your feedback. =)

That is one feedback i was waiting for when i thought about the functionality, but i wanted to wait for such feedback to this instead of setting it to a special way directly... 

Your idea is something i had in mind too, but if you agree i would even take it little further... What about another argument to querry the status of the text currently set on or off? I use several home automation environments on my smart mirror Raspberry Pi driven Magic Mirror central home automation controller thingi like HomeBridge, Node Red and ioBroker... Some of these would require 3 links. 2 for ON/OFF and a 3rd for the status... You alreday get the text reply if it set on or off but i think a status link would make it complete...

Let me know how you think about this and i will try to implement it soon... Was already on my list anyway for further improvements...

Thanks again and have a nice evening. :)

reptil1990, 2023-01-09:
Yes that would be ewen better so you can query the status directly and react if it was changed from Telegram or a different source. 
Then you could also syncronize different appliances according to a set ew status.

Yes i use ioBroker as well and now I have to keep track of the status myself. Would be nice to query it directly.

I already looked into the methods on the WebServer.h and the arguments can be easy used by the `String argName(int i); ` 

When I can get the code to compile on my main PC I also will look to help you out and make the changes and make a PR.
Depending on my time!

Thank you very much!

AWSW-de, 2023-01-09:
Sounds good!

As i typed the answer, I think the link can be even shorter:
- http://192.168.1.102:8080/ew1?1
- http://192.168.1.102:8080/ew1?0
- http://192.168.1.102:8080/ew1?s
but this would be just a cosmetic detail...

The code needs tons of cleanup... As you may have seen i just used many copy&paste actions just to get it ready first, but this is actually a bit of work just to add one more language... Just working on a French layout... With more dynamic functions used this will be way easier....

Take your time. This should be a fun project and if it grows over time i would like this very much... :)

You're welcome and thanks in advance!

reptil1990, 2023-01-10:
I would it not make as short as you said. Make it a little more clear.
So you can grab the arguments by name. Would make it better to implement more features. Like Brightness. Colors.

http://192.168.1.102:8080/ew1?on=1&br=100 -> Turn on and set brightness to 100% (make it percent not everybody will have the same range)

http://192.168.1.102:8080/ew1?cl=FFFFFF -> set color of the ew1 command to white

I know there is so much more but I think first step is the  and off. But a usefull schema to beginn with is better than changing it all the time.

Cheers

reptil1990, 2023-01-12:
I am not sure about that. 
You could take the current color value for the word and calculate the brightness for each word by taking the current value and run it through this funktion.

Here the funktion to do that:
`int decrease_brightness(int color, float percent) {
    int r = (color & 0xFF0000) >> 16;
    int g = (color & 0x00FF00) >> 8;
    int b = (color & 0x0000FF);
    r = (int)(r * percent);
    g = (int)(g * percent);
    b = (int)(b * percent);
    return (r << 16) | (g << 8) | b;`

Hey I don´t want to overstress your time.
Let´s start with on and off and then the future will bring the rest.

Thank you!

AWSW-de, 2023-01-12:
Interesting idea… Need to think about this… 
Another idea for such an enhancement would be to let the texts blink… This might be a good idea to make them even more visible….

AWSW-de, 2023-03-09:
Please see: https://www.printables.com/de/model/331012-wordclock-16x16-led-matrix 

09.03.2023:  Working on version V1.4.0:

    Enhanced internal web URL usage to be able to set/get the following values from your smart home environment systems that can handle such URLs:
        Change time color 
        Change background color
        Change intensity value (0 to the maximum limit set in the settings)
        Set extra words color
        Set extra words on or off
        Get the status of an extra word
        Get the status of all above named values (colors and on/off value) at once
        Examples: 
            http://WordClock:2023/?R-Time=255&G-Time=255&B-Time=255&R-Back=0&G-Back=0&B-Back=255&INTENSITY=25 → Cange time color to white, background to blue and intensity to 25
            http://WordClock:2023/?INTENSITY=25 → sets intensity to 25 
            http://WordClock:2023/?INTENSITY=0 → sets the display to off
            http://WordClock:2023/status → gets all current values to use them in your smart home systems like “R-Time=255 G-Time=128 B-Time=0 R-Back=0 G-Back=255 B-Back=255 INTENSITY=15 ew1=1 ew2=0 ew3=0 ew4=0 ew5=0 ew6=0 ew7=0 ew8=0 ew9=1 ew10=0 ew11=0 ew12=0”
            http://WordClock:2023/ew?5=1&R=255&G=255&B=255 → sets extra word 5 on and to white
            http://WordClock:2023/ew?3=0 → sets extra word 3 to off
            NOTE: Links may change a bit… This should just give you the idea…
    Enhanced functions for the manual Telegram interactions including enhanced startup message to get faster used to the usage.
    Enhanced startup outputs on the serial monitor for 1st usage
    MQTT support to control the device with this protocol as well
    (If possible: Trying to get rid of the WiFI manager to implement something better)
    Tons of code cleanup to optimise the code…

reptil1990, 2023-03-09:
Absolutly awsome! if you need a beta tester i am glad to help!

AWSW-de, 2023-03-09:
Thanks. I will let you know =)

AWSW-de, 2023-03-18:
V1.4.0 adds the required functions. Closed.

closed 2023-03-18 by AWSW-de