AWSW-de/WordClock-16x16-LED-matrix #2: Struggling to compile arduino code

remy, 2023-01-11:
I've run into a few problems trying to compile and I thought I could work around them but I've gotten to the point where I think I'm definitely doing something wrong.

Just to also say, I've downloaded and added the libraries listed at the top of the comments in the Code.ino file.

The first blocker was the build complaining that ESPUI couldn't find LittleFS regardless of how I installed the library. I eventually decided to check the build dates on the world clock project and the dates on ESPUI and I can see the latest ESPUI came out, specifically the commits between 2.1.1 and 2.2.0 include "Several LittleFS changes, fixes https://github.com/s00500/ESPUI/issues/192, adds option to not format LittleFS" - so I decided to install version 2.1.1 which came out in May 2022. Now the build gets past this point. 

However, now I'm stuck with these errors (I've shortened some of the file paths to make it easier to read):

```
Code:585:123: error: no matching function for call to 'ESPUIClass::button(const char [25], void (*)(Control*, int, void*), ControlColor, const char [25], void*)'
   ESPUI.button("Reset WordClock settings", &buttonWordClockReset, ControlColor::Dark, "Reset WordClock settings", (void*)3);
                                                                                                                           ^
In file included from Code.ino:48:
libraries/ESPUI-2.1.1/src/ESPUI.h:250:14: note: candidate: 'uint16_t ESPUIClass::button(const char*, void (*)(Control*, int), ControlColor, const String&)'
     uint16_t button(const char* label, void (*callback)(Control*, int), ControlColor color,
              ^~~~~~
libraries/ESPUI-2.1.1/src/ESPUI.h:250:14: note:   candidate expects 4 arguments, 5 provided
```

This repeats for every time that ESPUI::button is called. I can see the source of ESPUI's button method expects 4 args and your code is calling with 5. I arbitrarily tried removing the `(void*)3` but I got a new error which I didn't understand and commented the whole line out - but of course ESPUI::button is called a lot (and I'm assuming for good reason), so I didn't want to mess with the code too much, as I'm sure I've just got something wrong on my side.

Any ideas?

AWSW-de, 2023-01-11:
Hello,
please check when you downloaded the libraries from the pages directly and unpacked them that the suffix “-master” will be removed from the created folder and that the folder then contains the lib files directly…. Sometimes the unzip action creates the same folder as subfolder again and then the lib will not be found. Then just copy the folders to your “/UserProfile/Arduino/libraries/“ folder. Restart Arduino and that should do it I think. 

In addition please check that the verbose output is not activated in the Arduino preferences. 

Hope this helps

remy, 2023-01-11:
I'll give that a try, but I don't believe the directory name actually has any impact, as the Arduino app is reading each directory looking for `library.json` (and/or `library.properties`) which has the named path.

For example, this is the contents of `libraries/ESPUI-2.1.1/library.json`:

[image]

What do you have in the contents of your ESPUI/library.json?

remy, 2023-01-11:
(just to confirm, I had the same compiler warning settings as you too - set to `none`)

AWSW-de, 2023-01-11:
Strange... Except the version number it looks similar:
`{
  "name": "ESPUI",
  "keywords": "espressif web interface iot easy ui",
  "description": "ESP32 and ESP8266 Web Interface Library",
  "repository": {
    "type": "git",
    "url": "https://github.com/s00500/ESPUI.git"
  },
  "authors": [
    {
      "name": "Lukas Bachschwell",
      "email": "lukas@lbsfilm.at",
      "url": "https://lbsfilm.at",
      "maintainer": true
    }
  ],
  "dependencies": [
    {
      "name": "ESP Async WebServer",
      "authors": "Hristo Gochkov",
      "frameworks": "arduino"
    },
    {
      "name": "ArduinoJson",
      "authors": "Benoit Blanchon",
      "frameworks": "arduino"
    },
    {
      "name": "LittleFS_esp32",
      "authors": "lorol",
      "frameworks": "arduino"
    }
  ],
  "version": "2.2.1",
  "frameworks": "arduino",
  "platforms": "*"
}
`

remy, 2023-01-12:
Right, got to the bottom of it. The Arduino IDE had selected a _different_ (I think) ESP32 library.

[image]

The UI is _slightly_ different in IDE 2 (which I only found last night) and I can see when the board is selecting the different core (I think that's the right word?) - so now when I selected "esp32" version instead of "ESP32 Arduino" it builds.

I also added this url to my board list - not sure if had a difference or not: `https://dl.espressif.com/dl/package_esp32_index.json` (which I'd missed from your directions on the printables web site - when you link to how to upload).

(I've since also changed the board selection to "WEMOS D1 Mini esp32").

The "LittleFS" issue seems to be a bit of a problem for others using the ESPUI library (I found a few open issues on their github repo) - which it turns out is entirely down to the different ESP32 library board library being used.

Thanks for a great project - it's now up and running and looks great 👍

AWSW-de, 2023-01-12:
Thanks for your reply. Interesting what effects you can get through such small changes... I was thinking to deliver a web flash option to just upload a .BIN file to your ESP32 but this has a lot of other downsides then and can cause a lot of extra work need to be done, so that it might not be easier to install in the end...

Nice, that it works now and i wish you lots of fun with it. If you have a good idea to this project, feel free to let me know. :)

You're welcome and thanks for your nice words.