AWSW-de/WordClock-16x16-LED-matrix #3: Receiving tons of errors when compiling in IDE

thorsten0909, 2023-03-04:
I am constantly receiving multiple errors when trying to compile in Arduino IDE.
What am I doing wrong?

[errors.txt](https://github.com/AWSW-de/WordClock-16x16-LED-matrix/files/10888524/errors.txt)

AWSW-de, 2023-03-04:
Hi, 
looks like you copied the code from the web page here. There seem to be a lot of html commands there that should not be in the code… 

Please try the following:

- Download the code as .ZIP archive via the “green button” here: https://github.com/AWSW-de/WordClock-16x16-LED-matrix
- Unpack the downloaded .ZIP archive to a folder on your computer. 
- Make sure you find the “Code.ino” and “Settings.h” file in the folder.
- After Arduino IDE install (you have that already) open the code.ino file with the program
- You will need to add the following libraries to your Arduino IDE to use the project:
// # - Adafruit NeoPixel      // by Adafruit:                     https://github.com/adafruit/Adafruit_NeoPixel
// # - WiFiManager            // by tablatronix / tzapu:          https://github.com/tzapu/WiFiManager
// # - AsyncTCP               // by me-no-dev:                    https://github.com/me-no-dev/AsyncTCP
// # - ESPAsyncWebServer      // by me-no-dev:                    https://github.com/me-no-dev/ESPAsyncWebServer
// # - ESPUI                  // by s00500:                       https://github.com/s00500/ESPUI
// # - ArduinoJson            // by bblanchon:                    https://github.com/bblanchon/ArduinoJson
// # - LITTLEFS               // by lorol:                        https://github.com/lorol/LITTLEFS
// # - UniversalTelegramBot   // by witnessmenow:                 https://github.com/witnessmenow/Universal-Arduino-Telegram-Bot
- Do not use other libraries please to make your life more easy. Download again the ZIPs with the green buttons again… =)
- Copy the unZIPed files to: “C:\Benutzer\DeinUsername\Dokumente\Arduino\libraries” as separate folder like they were unZIPed with the extracted files in the single folders.
- Restart Arduino IDE
- Make sure that the board is “D1 Mini ESP32” —> If you don’t know how to do this, please follow [this link](https://randomnerdtutorials.com/installing-the-esp32-board-in-arduino-ide-windows-instructions/) and just select the ‘D1 mini ESP32” instead of the “DOIT ESP32 DEVKIT V1”… The rest should be almost the same…
- Compile and upload the code to the board…

HTH =)

AWSW-de, 2023-03-09:
Closes due to no response

closed 2023-03-09 by AWSW-de