AWSW-de/WordClock-16x16-LED-matrix #4: ESP32 not responsive

Spybyte, 2023-03-25:
Hi,
First of all, great project. I really love it. Thanks.

I had some issues with the code. I wasn't really able to connect to my wifi. But when there was a connection, the HTTP server didn't response. Pinging the ESP32 took up to 5 seconds or more.

Request timeout for icmp_seq 73
64 bytes from 192.168.1.81: icmp_seq=69 ttl=255 time=5140.429 ms
64 bytes from 192.168.1.81: icmp_seq=70 ttl=255 time=4140.202 ms
64 bytes from 192.168.1.81: icmp_seq=71 ttl=255 time=3136.441 ms

What I did was to add da delay of 500ms to the loop:

```
void loop() {
  delay(500);
  printLocalTime();
  ...
}
```

With this change, the ping time is acceptable:

64 bytes from 192.168.1.81: icmp_seq=0 ttl=255 time=117.110 ms
64 bytes from 192.168.1.81: icmp_seq=1 ttl=255 time=254.531 ms
64 bytes from 192.168.1.81: icmp_seq=2 ttl=255 time=106.225 ms
64 bytes from 192.168.1.81: icmp_seq=3 ttl=255 time=123.205 ms

That also fixed several issues for me:

- more reliable connection to my wifi
- faster start up
- responsive web server

I did not investigate further, but I think it could be the fast calls to `dnsServer.processNextRequest()` in the loop.

Maybe this helps.

AWSW-de, 2023-03-26:
Hello @Spybyte,
thanks a lot and thanks for the input. :) 

Really strange, because i run now around 15 of these devices here in the different languages and have not noticed such issues... 

Can you please give me some more details on your setup llike the used ESP32 board and did you try another one to compare this behavior? Is the power supply 5V/3A and how long is your USB cable... It could be as well that insufficiant power might let the chip strugle to work...

Thanks in advance :)

Spybyte, 2023-03-26:
Hello @AWSW-de ,

I‘m using a AZDelivery ESP32 D1 Mini NodeMCU board (the one you linkend on printables). Right now, it‘s just the plain board without the led matrix. So it‘s basically only attached to my computer with a cable length of 0.8m. 

So I tried to keep it simple before attaching the led matrix. 🙃

AWSW-de, 2023-03-26:
Hi @Spybyte,

really strange then… Can you try another board if you have a spare?

I have serval of these devices here running and the web interface is really fast possible to open with under 1 second….

But as it is working for you as i understand this might not be a common behaviour… I will keep that in mind and think about it for the next version….

Thanks and kind regards =)

closed 2023-03-26 by AWSW-de

Spybyte, 2023-03-27:
@AWSW-de I tested an older AZDelivery ESP32 NodeMCU D1 R32 board (not the mini). That worked without the delay. So I just ordered a second mini board from AZDelivery. I will let you know, if it has the same problem.

Cheers

Spybyte, 2023-03-29:
@AWSW-de Just one final comment on this: I just got another AZDelivery ESP32 D1 Mini NodeMCU board and do have the same problem with the responsiceness. As soon as I add the delay or remove dnsServer.processNextRequest() from the main loop, it works. So it might be that there is a problem in one of the libs used. Maybe because a use a new version. So a list of used lib versions from your side would be great.

AWSW-de, 2023-03-29:
Hi,
really strange… I am using the current libs listed in the top of the code… Arduino does update them on every start. Just installed a new device a few days ago… I will give the hint a try to see if it speeds it even more… Thanks