AWSW-de/WordClock-16x16-LED-matrix #5: Integration of rtc module

thorsten0909, 2023-06-30:
Hi there, great project!
Is there a chance to use an rtc module like DS3231 I2C with this?
I would love to use the wordclock at work but have no access to wifi there...

AWSW-de, 2023-06-30:
Hello and thanks for the nice words!

Hmmm 🤔 I could integrate one like in my 1st WordClock, but to be honest this does not solve the whole problem in my eyes… You would loose the whole configuration function and would not be able to set colours etc… Of course the whole smart functions would be missing too…

I have a different idea since some time for such requests… Maybe you can think about the following idea too to find a better solution without the need of extra hardware, because I had some issues with these RTC boards in the past and the ESP32 can solve this alone I think…

My approach would be to have a toggle switch in the software to switch the “always online functions all to off” because you have no WiFi and do not want to use an extra RTC…

When this switch is set to on the whole WiFi stuff is set to off and all of the smart functions too of course… Then you would be able to connect to the board via its Captive Portal function which is used to setup the initial WiFi but not to configure the WiFi but to enter the “offline possible configurations” like colours etc and to set the current time manually… from that on the WordClock would work offline and you always can enter the offline possible configuration options with the web portal too…

Hopefully you can understand my idea? 

That would be something I would like to follow because an offline WordClock should be possible to configure anyway for the then still possible functions in my eyes…

Please let me know what you think about it. 😊

Many regards =)

AWSW-de, 2023-06-30:
This is the current online variant with WiFi:
[image]
[image]
[image]
[image]

AWSW-de, 2023-06-30:
and this could be the „offline variant“ (without the WiFi reset button of course)
[image]
[image]

thorsten0909, 2023-06-30:
This sounds like an awesome but still simple solution and would do the trick for me.

I just need a nice piece of art on my office desk, to show the time.

Since it is not in its smart home environment anyway, I would not be able to use the smart functions after all.

Just access through the AP mode (could it be “always on”?) to set time and colors would be sufficient.

AWSW-de, 2023-06-30:
Thanks for the reply

Yes, I made a „CO2 traffic light” / “CO2 Ampel” some time ago which kept such an captive portal web server all the time to view the values all the time… Should work, but I need to think about this… 

This will definitely take some time but looks like a nice challenge to solve... 😁

thorsten0909, 2023-06-30:
Danke dir vielmals für die Bereitschaft, dir das anzuschauen!Thorsten RichterSent from a mobile device - please excuse typing mistakes.This communication contains confidential information. If you are not the intended recipient please return this email to the sender and delete it from your records.Diese Nachricht enthaelt vertrauliche Informationen. Sollten Sie nicht der beabsichtigte Empfaenger dieser E-mail sein, senden Sie bitte diese an den Absender zurueck und loeschen Sie die E-mail aus Ihrem System.

AWSW-de, 2023-06-30:
Kein Ding, gerne 👍

AWSW-de, 2023-07-02:
02.07.2023: Working on the BIGGEST update (V2.0.0) yet =)

- Replacing the old WiFi manager library with my own solution already used in the 2023 models to avoid many problems users reported and to have more control.

- This new WiFi and config manager also allows to set the NTP time server, time zone, language of the LED layout and Telegram settings. These values now get stored in the ESP32 flash and does remove the amount of effort to change the settings with every new update in the settings.h file again and again… —> Done =)

- More control and LED output during the setup process. —> Done =)

- New internal web pages for the initial setup and update process. —> Done =)

- Next is your request… Keep fingers crossed 🤞 —> Ongoing… =)

AWSW-de, 2023-07-06:
Made some really good progress on this yesterday. Needs some more optic finishing but the general functions work as expected and I will add some more special features to this “Offline Mode” called operation mode which I will explain later.  Keep fingers crossed 🤞

AWSW-de, 2023-07-06:
Hello,
software version V2.0.0 is online now. Please update and let me know what you think. :)

WordClock can be used in "Online Mode" with all the smart functions controlled via your network environment and otional via Telegram and also in "Offline Mode" where you can use the time piece function only without the need of a local WiFi. In both operation modes you can configure the device via the shown internal web portal.

closed 2023-07-06 by AWSW-de