AWSW-de/WordClock-16x16-LED-matrix #7: Error on Letter

valbuz, 2024-01-02:
Hi
Does anyone know this error?

thanks
pat

[image]

AWSW-de, 2024-01-02:
Hello and happy new year,

can you please explain what error? Do you mean the 2 „H“ pixels in red color? Hard to see on the picture…

In case you mean the 2 red letters. This can be everything from:
- Insufficient power source —> You should use the named  5V / 3A power supply.
- Too less power because of to high led brightness intensity. —> Lower intensity and test again
- Dead pixels in the led matrix ? —> try another sketch to test the function, but take care with the brightness settings! You can easily kill the ESP32 with to high settings like described in the settings.h file… Don’t go higher than 50 and do not use white background settings. 

Hope to help. 

PS: Und nun sehe ich auch erst, dass ich Dir wahrscheinlich auch in deutsch hätte schreiben können… 😂🙈

Kind Regards / Viele Grüße 🖖

valbuz, 2024-01-02:
Deutsch ist auch gut :-)
Danke für die Antwort und ebenfalls E guets Neus.
Sorry, ja es ist erst ein Prototyp darum vielleicht nicht die optimale Einstellung betr. Backlight.
Also Matrix scheint OK zu sein.
Vermutlich an der Speisung ;-)
Danke nochmals, Super Projekt!
P.S. Vielleicht wäre für die Zukunft MQTT noch etwas?
Gruss
Patrick

AWSW-de, 2024-01-02:
Hi, 
ja, vermutlich eine Unterversorgung. 

MQTT hatte ich mal auf die Liste gesetzt aber wieder verworfen, da der Webserver eh läuft und man Alles über die Links steuern kann. Der ESP32 hat ja auch nur sehr begrenzte Ressourcen….

Viel Spaß mit der WordClock 😁