AWSW-de/WordClock-16x16-LED-matrix #9: Fehlermeldung beim Installieren

Plampi20, 2024-04-28:
Hallo was kann ich machen, das diese Meldung nicht mehr kommt, und ich alles Installieren kann um die Uhr zu Starten.
Fehlermeldung ist folgende:
C:\Users\19pla\Downloads\AWSW-WordClock-Code\WordClock-16x16-LED-matrix-main\Code\Code.ino:47:10: fatal error: Adafruit_NeoPixel.h: No such file or directory
 #include <Adafruit_NeoPixel.h>     // Used to drive the NeoPixel LEDs
          ^~~~~~~~~~~~~~~~~~~~~
compilation terminated.
exit status 1

Compilation error: Adafruit_NeoPixel.h: No such file or directory

Ich habe alle Dateien in den Libarie im Arduino PLC IDE ordner abgelegt.

[image]

AWSW-de, 2024-04-28:
Hi, 
die Libraries, die oben in der Code.ino als Links angegeben sind, scheinen zu fehlen. Zur Installation siehe: https://youtu.be/BrVKY1sfZgI?si=E-b_kccjOUfWDsNv 
oder dieses Skript zur automatischen Installation nutzen: https://github.com/AWSW-de/AWSW-WordClock-Windows-setup-script

Viele Grüße 🖖

closed 2024-04-28 by AWSW-de

Plampi20, 2024-04-28:
Ich habe das Skript zur automatischen Installation benutzt.

AWSW-de, 2024-04-28:
In dem Fall liegen die Libraries schon im richtigen Ordner und müssen nicht in einen anderen Ordner gelegt werden. Bitte das Video noch mal schauen und dann ggf. noch mal das Skript ausführen. Einzig, wie im Video gezeigt. müssen die ESP32 Boardtreiber noch installiert werden. Dann funktioniert es. Viel Erfolg.

AWSW-de, 2024-04-28:
Gerade noch mal aktuell geprüft und kappt 1a: 

[image]