AWSW-de/WordClock-16x16-LED-matrix #10: Fehlermeldung "Sketch is too big"

Napizaki, 2024-05-31:
Hallo,

wenn ich, wie in der Videoanleitung beschrieben, die V3.80 für einen WEMOS D1 MINI ESP32 kompilieren möchte, bekomme ich folgende Fehlermeldung:

[image]

Alle Libraries sind die derzeit aktuellen.

AWSW-de, 2024-05-31:
Hallo,
ich habe das gerade mit komplett neuer Installation probiert mit dem Skript (https://github.com/AWSW-de/AWSW-WordClock-Windows-setup-script) installiert und dann nur noch die ESP32 Boards installiert:

[image]

Damit konnte ich das Verhalten nachvollziehen. Das Problem scheint die neue Version 3.0.0 der ESP32 Boards von Expressive zu sein. Bitte im Bordverwalter die ESP32 Board Version von 3.0.0 wieder auf 2.0.17 stellen und somit downgraden:

[image]

[image]

Denke, die ESP32 V3.0.0 hat noch einen Bug. Daher bitte die vorherige Version verwenden und dann kopiliert der Code wieder:

[image]

Viel Spaß mit der WordClock. =)

Viele Grüße
AWSW

AWSW-de, 2024-05-31:
Habe die FAQs für die 16x16 und 16x8 WordClocks erweitert

Napizaki, 2024-05-31:
Perfekt, hat funktioniert. Danke für die schnelle Hilfe!!!

AWSW-de, 2024-05-31:
Alles klar, danke fürs Feedback. Ich schließe das Ticket und hoffe, der Bug wird bald behoben.

closed 2024-05-31 by AWSW-de

AWSW-de, 2024-06-01:
Moin, na wenigstens liegt es nicht an meinem Code, welcher in verschiedenen Projekten nun das gleiche Verhalten zeigt.
Wurde bereits gemeldet https://github.com/espressif/arduino-esp32/issues/9741 und mal sehen wann es behoben wird...