AWSW-de/WordClock-16x16-LED-matrix #11: Constant reboot after getting internettime

Napizaki, 2024-08-15:
Hello,

I have built myself a 2nd watch. But after installing the software and configuring the WiFi, the clock connects to the WiFi and synchronizes the time. The clock should then display the time, but instead it keeps restarting - always at the same point.
The libraries are all the latest ones (except for the ESP32 boards - version 3.0.3 still doesn't work there - here I use 2.0.17).
I then took my 1st clock and installed the new software on it to rule out the possibility that it was due to the hardware - this also constantly restarts with the new software - also in the same point as described above.
I also tried all possible cables and power supply units. My 1st watch had also been running for several months without any problems.
There must be something wrong with the new software (or the libraries).

Tanks for the help in advance!

AWSW-de, 2024-08-15:
Hello,

please make sure that you have not updated the ESPUI library higher than 2.2.3. The newest version 2.2.4 I think is the actual version and did not work on ESP32 in my experience. 

Hope to help. 

Kind regards and enjoy your WordClock =)

Napizaki, 2024-08-15:
Right now I saw it as well - and It works fine now.

[image]

Thank you for the fast response!

AWSW-de, 2024-08-15:
OK nice. Enjoy your WordClock =)

closed 2024-08-15 by AWSW-de