AWSW-de/WordClock-16x16-LED-matrix #13: Size issue

crypto-7oM, 2024-10-15:
Hello,
I'm very new to this and I'm having a heck of a time trying to get this uploaded to my ESP32. I have the DOIT ESP32 Devkit 1, according to specs, it has 4MB of flash memory.  I keep getting an error that the sketch size is 101% and too big for the partition.  I've tried multiple times to re-partition to make room, but it doesn't seem to work.

In the instructional video, it shows that the file when uploaded via Arduino IDE is about 700k, whereas my file seems to be 1.3MB.  Any idea how I can trim this file down or omit some things to make it smaller? I tried getting rid of comments, but that really didn't do anything.

Thanks for any help!

AWSW-de, 2024-10-15:
Hello,
you simply using the wrong version of the ESP32 board library. The one that will work is V2.0.17. Via update you may have updated to V3.0.x in the Arduino IDE. Arduino has made some changes that limits the space.  

Downgrade the ESP32 board library to 2.0.17 and make sure to use exactly the names libraries on top of the code. Then it will upload. 

About the board I cannot say anything. I would recommend the named ESP32 for this project to work correctly. 

If you’re not familiar with setting up this environment you can use this script to handle this task for you. https://github.com/AWSW-de/AWSW-WordClock-Windows-setup-script

Enjoy your WordClock =)

crypto-7oM, 2024-10-15:
Thank you so much! I got it working.  I tried to downgrade the board library myself but it still wasn't working. I let the script to its thing and it worked after setting the pin.

Is this script in the github? I must have missed it.  Thanks again!

closed 2024-10-15 by crypto-7oM

AWSW-de, 2024-10-15:
Hello, nice to hear. 
Yes it is available here too.
Enjoy your WordClock =)
Kind regards