AWSW-de/WordClock-16x16-LED-matrix #14: Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

GimmeMe1, 2024-10-20:
Hello
After configuring WordClock it's rebooting always because of Core 1 panic'ed error. Can you help me to fix this?

######################################################################
# WordClock startup of version: V4.0.0
######################################################################
 
Try to connect to found WiFi configuration: 
Connecting to WiFi HA
Connection try #: 1 of maximum 100
 
Successfully connected now to WiFi SSID: HA
IP: 192.168.1.81
DNS: 192.168.1.1
GW: 192.168.1.1
ESP32 hostname: esp32-E4932C
Setting up time from: pool.ntp.org
Got the time from NTP server: pool.ntp.org
Setting timezone to CET-1CEST,M3.5.0,M10.5.0/3
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x400f4bac  PS      : 0x00060130  A0      : 0x800f4cc2  A1      : 0x3ffb1fe0  
A2      : 0x3ffc4c9c  A3      : 0x00000000  A4      : 0x3f403401  A5      : 0x3ffb201c  
A6      : 0x00000006  A7      : 0x0000ffff  A8      : 0x800f4bac  A9      : 0x3ffb1fb0  
A10     : 0x3ffc4cd8  A11     : 0x00000002  A12     : 0x00000000  A13     : 0x00000000  
A14     : 0x3ffb7e24  A15     : 0x00000000  SAR     : 0x00000017  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000002  LBEG    : 0x4008a5f0  LEND    : 0x4008a606  LCOUNT  : 0xffffffff  

Backtrace: 0x400f4ba9:0x3ffb1fe0 0x400f4cbf:0x3ffb2000 0x400dfaa2:0x3ffb2050 0x400e4647:0x3ffb2170 0x400e4bd1:0x3ffb2230 0x401007b6:0x3ffb2290

ELF file SHA256: e10259734905703e

Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1184
load:0x40078000,len:13260
load:0x40080400,len:3028
entry 0x400805e4
 
 thank you!

Best 

Thomas

AWSW-de, 2024-10-21:
Hello Thomas,

looks like you’re using the newer versions of the libraries which are not compatible in their new versions anymore. You need to downgrade them as written in the FAQ on printables or you just use this script https://github.com/AWSW-de/AWSW-WordClock-Windows-setup-script which will clean all your downloads regarding the WordClock code and will install the Arduino IDE new with the required versions of the libraries. Then it will the easy to upload and use the device. Details to the script is shown in the video. 

Enjoy your WordClock =)

closed 2024-10-21 by AWSW-de

GimmeMe1, 2024-10-21:
Hello
actually i used that installation script, which should install the correct version of libraries. So I don't think this is the problem to address. Any other ideas?
Best Thomas

AWSW-de, 2024-10-21:
Hi,

you need to make sure that then auto update of the Arduino IDE does not update them on the start which is now turned of with the V2.x.y script. 

So now I would check that the 

1. ESP32 board driver library is 2.0.17
2. The ESPUI library is 2.2.3

That should help. If not, I would try to use another ESP32 or upload another sketch to see if it works in general.

Hope to help =)

GimmeMe1, 2024-10-21:
Hello
ESPUI 2.2.4 was installed although i used the installation V2.0.1 script. Now with 2.2.3 it's working. Perfect and thank you.
Best Thomas